jvcleave/ofxOMXPlayer #144: Draw a non rectangle shape

oomek, 2019-03-20:
Hi, I have several questions if I may:
1. Is it possible to draw the video window in non rectangular (perspective) shape providing 4 pairs of coordinates?
2. Can I integrate it with sfml-pi so the draw order is preserved? ie. draw some sfml objects, call ofApp.draw() and then draw the rest of sfml objects?
3. Is it possible somehow to get a GLuint of the currently decoded frame so I could draw it with Opengl with zero copy?

jvcleave, 2019-03-20:
1. yes with texture mode
2. not familiar with sfml-pi
3. yes in texture mode see https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayer.h#L54

oomek, 2019-03-20:
That's great, If only I could find a way to integrate it with SFML :P  not sure if it's actually possible

oomek, 2019-03-20:
I'm not gonna lie t'm actually looking for a conventional c++ wrapper for the omxPlayer which would give me an access to the Gluint of the rendered frame as the current ffmpeg+mmal method we are using in our app based on the SFML framework has dreadful performance.

oomek, 2019-03-20:
I'll dig through your code for inspiration maybe if you don't mind.

jvcleave, 2019-03-20:
good luck - you can also check out hello_videocube if you want an alternative route

https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi/hello_videocube

closed 2019-03-20 by jvcleave

oomek, 2019-03-20:
Thanks for the link. Unfortunately this example plays only raw h264 files without audio.