jvcleave/ofxOMXPlayer #146: Segfault while creating or destructing ofxOMXPlayer objects

CyrCom, 2019-03-28:
Hello,

I stumble upon an unpredictable segmentation fault running my program.
To reproduce the crash:
I create and I destruct a lot of ofxOMXPlayer objects, and the crash can happen in one or the other case... It means that it is usually fine, but sometimes, it is crashing.
I'm not doing anything to the object between creation and destruction.
I might not be ofxOMXPlayer's fault, but you might be aware of something.
Here is the GDB backtrace

```
Thread 1 "omxTCPWarp_debu" received signal SIGSEGV, Segmentation fault.
0x0010a308 in ofxOMXPlayer::onUpdate (this=0x2242c58, eventArgs=...) at ../../../addons/ofxOMXPlayer/src/ofxOMXPlayer.cpp:326
326	                listener->onVideoLoop(this);
#0  0x0010a308 in ofxOMXPlayer::onUpdate (this=0x2242c58, eventArgs=...) at ../../../addons/ofxOMXPlayer/src/ofxOMXPlayer.cpp:326
#1  0x0010c524 in std::shared_ptr<of::priv::Function<ofEventArgs, std::recursive_mutex> > ofEvent<ofEventArgs, std::recursive_mutex>::make_function<ofxOMXPlayer>(ofxOMXPlayer*, void (ofxOMXPlayer::*)(ofEventArgs&), int)::{lambda(void const*, ofEventArgs&)#1}::operator()(void const*, ofEventArgs&) const (__closure=0x2207378, t=...) at ../../../libs/openFrameworks/events/ofEvent.h:479
#2  0x0010e1ec in std::_Function_handler<bool (void const*, ofEventArgs&), std::shared_ptr<of::priv::Function<ofEventArgs, std::recursive_mutex> > ofEvent<ofEventArgs, std::recursive_mutex>::make_function<ofxOMXPlayer>(ofxOMXPlayer*, void (ofxOMXPlayer::*)(ofEventArgs&), int)::{lambda(void const*, ofEventArgs&)#1}>::_M_invoke(std::_Any_data const&, void const*&&, ofEventArgs&) (__functor=..., 
    __args#0=<unknown type in /home/pi/of_v20180811_linuxarmv6l_release/apps/omxTCPWarp/omxTCPWarp/bin/omxTCPWarp_debug, CU 0x181666, DIE 0x19e08b>, __args#1=...) at /usr/include/c++/6/functional:1717
#3  0x0014d14c in std::function<bool (void const*, ofEventArgs&)>::operator()(void const*, ofEventArgs&) const (this=0x22498ac, __args#0=0x0, __args#1=...) at /usr/include/c++/6/functional:2127
#4  0x0014a2b4 in of::priv::Function<ofEventArgs, std::recursive_mutex>::notify (this=0x22498a4, s=0x0, t=...) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/events/ofEvent.h:91
#5  0x00148100 in ofEvent<ofEventArgs, std::recursive_mutex>::notify (this=0x1b108d0, param=...) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/events/ofEvent.h:578
#6  0x001471ac in ofNotifyEvent<ofEvent<ofEventArgs, std::recursive_mutex>, ofEventArgs> (event=..., args=...) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/events/ofEventUtils.h:214
#7  0x00144004 in ofCoreEvents::notifyUpdate (this=0x1b108c8) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/events/ofEvents.cpp:286
#8  0x001e2804 in ofAppEGLWindow::update (this=0x1b105a8) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/app/ofAppEGLWindow.cpp:860
#9  0x001fc378 in ofMainLoop::loopOnce (this=0x1b0e638) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/app/ofMainLoop.cpp:144
#10 0x001fc154 in ofMainLoop::loop (this=0x1b0e638) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/app/ofMainLoop.cpp:125
#11 0x001f7ec4 in ofRunMainLoop () at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/app/ofAppRunner.cpp:200
#12 0x001f7d94 in ofRunApp (OFSA=0x1b21f50) at /home/pi/of_v20180811_linuxarmv6l_release/libs/openFrameworks/app/ofAppRunner.cpp:177
#13 0x000d3e48 in main () at /home/pi/of_v20180811_linuxarmv6l_release/apps/omxTCPWarp/omxTCPWarp/src/main.cpp:8
```

it might lead there:

```
void ofxOMXPlayer::onVideoLoop(bool needsRestart)
{
    ofLogNotice(__func__) << endl;
    pendingLoopMessage = true;
    engineNeedsRestart = needsRestart;
}

void ofxOMXPlayer::onUpdate(ofEventArgs& eventArgs)
{
    if(engineNeedsRestart)
    {
        engineNeedsRestart = false;
        if(isOpen())
        {
            engine.close();
        }
        if(pendingLoopMessage)
        {
            if(listener)
            {
                listener->onVideoLoop(this);    // THIS IS LINE 326
            }
            pendingLoopMessage = false;
        }
        setup(settings);
        
        
    }else
    {
        if(pendingLoopMessage)
        {
            if(listener)
            {
                listener->onVideoLoop(this);
            }
            pendingLoopMessage = false;
        }
    }
    
}
```

Thanks for maintaining the project, it's dope

jvcleave, 2019-03-30:
have you tried setting the listener to null before destroying the ofOMXPlayer instance?

jvcleave, 2019-03-30:
you might have to set settings.listener = null as well if you are passing in the listener that way

CyrCom, 2019-04-06:
Hey,
Sorry for delay,
My commit resolves my problems.
Check it out, I think it's important to remove listeners on destruction (player + engine)