jvcleave/ofxOMXPlayer #150: Can't find ofxOMXPlayerListener.h

minizzZ, 2019-06-06:
I’m trying to install PocketVJ on a Raspberry Pi 3B+ and I get a fatal error about not finding ofxOMXPlayerListener.h. You can look at the installation for it [here](https://gist.github.com/baydam/a905e4878f54150f7da421e2c1d17a21). This is what I see in the terminal:
`In file included from /home/pi/openFrameworks/addons/ofxOMXPlayer/src/OMXAudio.h:30:0,
                 from /home/pi/openFrameworks/addons/ofxOMXPlayer/src/ofxOMXPlayerEngine.h:7,
                 from /home/pi/openFrameworks/addons/ofxOMXPlayer/src/ofxOMXPlayer.h:3,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/omPlayer.h:3,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/mediaPlayer.h:2,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/HPlayer.h:4,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/main.cpp:2:
/home/pi/openFrameworks/addons/ofxOMXPlayer/src/linux/PlatformDefs.h:172:0: warning: "APIENTRY" redefined
 #define APIENTRY    WINAPI
 
In file included from ../../../libs/openFrameworks/app/ofAppGLFWWindow.h:6:0,
                 from ../../../libs/openFrameworks/ofMain.h:80,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/main.cpp:1:
../../../libs/glfw/include/GLFW/glfw3.h:98:0: note: this is the location of the previous definition
   #define APIENTRY
 
In file included from /home/pi/openFrameworks/apps/myApps/HPlayer/src/mediaPlayer.h:2:0,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/HPlayer.h:4,
                 from /home/pi/openFrameworks/apps/myApps/HPlayer/src/main.cpp:2:
/home/pi/openFrameworks/apps/myApps/HPlayer/src/omPlayer.h:4:34: fatal error: ofxOMXPlayerListener.h: No such file or directory
 #include "ofxOMXPlayerListener.h"
                                  ^
compilation terminated.
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/main.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/main.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/HPlayer'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2`

I had these sort of errors before and than I would just nano the text of the .h and change where it could find the file. Now I looked my entire Raspberry even in the root but I can’t find ofxOMXPlayerListener.h. Even when I look on the internet I can’t find omxplayerlistener anywhere. Now I’m a bit of a noob when it comes to Raspberries so i even don’t know what a listener is. Can you help me?

jvcleave, 2019-06-06:
try deleting the line ` #include "ofxOMXPlayerListener.h"`

ofxOMXPlayerListener is defined in the file ofxOMXPlayer.h so it should be fine

minizzZ, 2019-06-06:
Thanks, that did the trick. NowI have another poco problem. I've changed the poco version like in [here](https://gist.github.com/baydam/ef2e37ea158eb49ad94bc5e5cec0520a) but now I get this:

`ofImage.cpp:(.text+0x237c): undefined reference to `Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofImage.o): In function `bool loadImage<unsigned short>(ofPixels_<unsigned short>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
ofImage.cpp:(.text+0x2830): undefined reference to `Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofImage.o): In function `bool loadImage<unsigned char>(ofPixels_<unsigned char>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
ofImage.cpp:(.text+0x2ce4): undefined reference to `Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofUtils.o): In function `ofLaunchBrowser(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)':
ofUtils.cpp:(.text+0x870c): undefined reference to `Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofUtils.cpp:(.text+0x8734): undefined reference to `Poco::URI::setQuery(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofUtils.cpp:(.text+0x8760): undefined reference to `Poco::URI::toString[abi:cxx11]() const'
ofUtils.cpp:(.text+0x8948): undefined reference to `Poco::URI::toString[abi:cxx11]() const'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofURLFileLoader.o): In function `ofURLFileLoaderImpl::threadedFunction()':
ofURLFileLoader.cpp:(.text+0x3c58): undefined reference to `Poco::Thread::setName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofURLFileLoader.cpp:(.text+0x53ac): undefined reference to `Poco::SystemException::SystemException(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
ofURLFileLoader.cpp:(.text+0x5430): undefined reference to `Poco::SystemException::SystemException(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofURLFileLoader.o): In function `ofURLFileLoaderImpl::handleRequest(ofHttpRequest)':
ofURLFileLoader.cpp:(.text+0x5e3c): undefined reference to `Poco::URI::URI(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofURLFileLoader.cpp:(.text+0x5e54): undefined reference to `Poco::URI::getPathAndQuery[abi:cxx11]() const'
ofURLFileLoader.cpp:(.text+0x5ea0): undefined reference to `Poco::Net::HTTPRequest::HTTPRequest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofURLFileLoader.cpp:(.text+0x5ec0): undefined reference to `Poco::Net::NameValueCollection::add(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofURLFileLoader.cpp:(.text+0x5f28): undefined reference to `Poco::Net::HTTPSClientSession::HTTPSClientSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short)'
ofURLFileLoader.cpp:(.text+0x6238): undefined reference to `Poco::Net::HTTPClientSession::HTTPClientSession(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned short)'
ofURLFileLoader.cpp:(.text+0x63f0): undefined reference to `Poco::Exception::displayText[abi:cxx11]() const'
ofURLFileLoader.cpp:(.text+0x6484): undefined reference to `Poco::Exception::displayText[abi:cxx11]() const'
ofURLFileLoader.cpp:(.text+0x66e8): undefined reference to `Poco::Net::HTTPRequest::HTTP_GET[abi:cxx11]'
ofURLFileLoader.cpp:(.text+0x66ec): undefined reference to `Poco::Net::HTTPMessage::HTTP_1_1[abi:cxx11]'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofURLFileLoader.o): In function `ofURLFileLoaderImpl::ofURLFileLoaderImpl()':
ofURLFileLoader.cpp:(.text+0x7208): undefined reference to `Poco::Net::Context::Context(Poco::Net::Context::Usage, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Poco::Net::Context::VerificationMode, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ofURLFileLoader.cpp:(.text+0x77bc): undefined reference to `Poco::Exception::displayText[abi:cxx11]() const'
ofURLFileLoader.cpp:(.text+0x791c): undefined reference to `Poco::Exception::displayText[abi:cxx11]() const'
../../../libs/openFrameworksCompiled/lib/linuxarmv6l/libopenFrameworks.a(ofURLFileLoader.o): In function `Poco::ScopedLock<Poco::FastMutex>::~ScopedLock()':
[... cut ...]
X509Certificate.cpp:(.text+0xe88): undefined reference to `sk_num'
X509Certificate.cpp:(.text+0xe9c): undefined reference to `sk_value'
collect2: error: ld returned 1 exit status
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:382: recipe for target 'bin/HPlayer' failed
make[1]: *** [bin/HPlayer] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/HPlayer'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
`

jvcleave, 2019-06-06:
sorry - no idea about Poco :/

closed 2019-06-06 by jvcleave