jvcleave/ofxOMXPlayer #154: can't get drawCropped to work

crx091081gb, 2019-10-14:
I don't seem to be able get drawCropped to produce anything but black output even when using the bunny mov.

jvcleave, 2019-10-14:
Do you have textures enabled? I am pretty sure drawCropped only works in direct mode

crx091081gb, 2019-10-14:
I'll try turning it off and working from the direct mode example.

jvcleave, 2019-10-14:
with texture mode you should be able to use drawSubsection

```
omxPlayer.getTextureReference().drawSubsection(0, 0, 100, 100);
```

jvcleave, 2019-10-14:
Didn't test the above but check this

https://openframeworks.cc//documentation/gl/ofTexture/#!show_drawSubsection

crx091081gb, 2019-10-14:
Thanks. Just got it working in the direct mode example.... and yup works in application! Thanks.

Might try the subsection tip but I think I can get away without needing textures, was just considering whether to to run it through a shader and resample it but that looks a lot easier.

closed 2019-10-14 by crx091081gb