jvcleave/ofxOMXPlayer #156: cannot compile under RasperryPi4/OF 0.11

Bencilari, 2019-12-28:
Made some test, but cannot compile this addons. Every shipped example compilation fails with
`/usr/bin/ld: //opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line`
The library is in the place were's expected.
The addons is downloaded from Releases, but i'm pretty sure that i've tested the branch, still without success
Some other informations:
- ofxOMXPlayer under OF/addons
- example-basic (or any other) as subdir of ofxOMXPlayer or under OF/apps/myApps/
- make

Thanks!

Bencilari, 2020-01-06:
I've got it to compile, adding `-lbrcmEGL` in `addon_config.mk` last line. So:
`ADDON_LDFLAGS= -lavcodec -lavformat -lswscale -lavutil -lswresample -lavfilter -lm -lz`
became:
`ADDON_LDFLAGS= -lavcodec -lavformat -lswscale -lavutil -lswresample -lavfilter -lm -lz -lbrcmEGL`
But the addons still doesn't work... hangup and need a sigkill to term...
I'll open another issue with the output i'm gettin

MichaelHazani, 2020-01-18:
same here, seconding the questions and problem.

jvcleave, 2020-01-27:
pushing all RPI4 issues to here:
https://github.com/jvcleave/ofxOMXPlayer/issues/157

closed 2020-01-27 by jvcleave