jvcleave/ofxOMXPlayer #157: Raspberry Pi 4 Support

jvcleave, 2020-01-27:
I don't believe this addon (at least texture mode) is going to be ported to the RPI4

From the [RPI forums](https://www.raspberrypi.org/forums/viewtopic.php?p=1597746&sid=e07b9bd7dedbfa3c0d319c8003289c6f#p1597746)

"If you mean OpenMAX egl_render component, it is not compatible with the Pi4 as it tries talking to the firmware GLES driver, which doesn't exist on the Pi4. I really ought to look at making the component create for that fail on Pi4. "

jvcleave, 2020-01-28:
Direct mode is working in this branch
https://github.com/jvcleave/ofxOMXPlayer/tree/rpi4-develop

see example-direct-mode

autr, 2020-07-30:
Just an update to say this is now on omxplayer repo 😢

"Note: omxplayer is being deprecated and resources are directed at improving vlc."

AurelienConil, 2024-10-07:
Hi there
I would like to be updated if there was any substitute to continue playing full HD video on openframeworks over RPI in 2024
Is https://github.com/Jonathhhan/ofxVLCVideoPlayer could be an option ?