jvcleave/ofxOMXPlayer #161: Problem compiling ofxOMXPlayer: undefined reference to symbol 'eglCreateImageKHR'

Dazzid, 2020-10-05:
I'm trying to run the ofxOMXPlayer in RaspBerry Pi 3 (debian version 10.6, Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux) with OF version 0.11.0.

After following the instructions, the compiler is having problems with 'eglCreateImageKHR' 
The message is the following:
/usr/bin/ld: /home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayerEngine.o: undefined reference to symbol 'eglCreateImageKHR'
/usr/bin/ld: //opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line

What could be the problem, perhaps I'm running an incompatible OF version?

jvcleave, 2020-10-05:
not sure about anything post-Raspian Stretch but this may be needed

https://github.com/jvcleave/ofxOMXPlayer/compare/rpi4-develop#diff-eb5061a0f330b6b75449499e7932f9c0R653

Dazzid, 2020-10-06:
hi, I also tried the rpi4 branch, but still have the same compiler error. Should I install an older Raspbian version?

Dazzid, 2021-09-14:
Using Raspian Stretch and RBpi3 B+ with OF_v10.0, I'm having an error. 

I might be missing something? Perhaps there is a specific Stretch version?

Message:
HOST_OS=Linux
checking pkg-config libraries: cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk±3.0
with PKG_CONFIG_LIBDIR=
[warning] ofAppEGLWindow: init(): X11 not availble on RPI yet, using a native window instead
[ error ] ofAppEGLWindow: getScreenSize(): tried to get display size but failed
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 0x0
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 0x0
[ error ] ofAppEGLWindow: setupRPiNativeWindow(): dispman_element == DISPMANX_INVALID
[warning] ofAppEGLWindow: getNativeDisplay(): window not initialized, returning NULL
[warning] ofAppEGLWindow: getNativeDisplay(): window not initialized, returning NULL
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display

failed to add service - already in use?
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:190: fallo en las instrucciones para el objetivo ‘run’
make: *** [run] Error 1