jvcleave/ofxOMXPlayer #165: Build errors due to include /bin/locale in 0.10.0 release

glitchcore, 2021-03-25:
I try to build ofxOMXPlayer

First of all I install openframeworks 0.10.0 by instructions from https://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/

```
sudo apt-get clean
sudo apt-get update
# expand filesystem and set GPU memory to 128

# installing openframeworks
wget https://openframeworks.cc/versions/v0.10.0/of_v0.10.0_linuxarmv6l_release.tar.gz
mkdir openFrameworks
tar vxfz of_v0.10.0_linuxarmv6l_release.tar.gz -C openFrameworks --strip-components 1
cd /home/pi/openFrameworks/scripts/linux/debian

nano install_dependencies.sh # change libgles1-mesa-dev to libglvnd-dev
sudo ./install_dependencies.sh 
make -j3 Release -C /home/pi/openFrameworks/libs/openFrameworksCompiled/project
cd /home/pi/openFrameworks/examples/graphics/polygonExample
nano config.make # add -latomic to linker script
make
make run

# successful running plygon example
```

Then I try to build 0.10.0 release:

```
cd /home/pi/openFrameworks/addons/
wget https://github.com/jvcleave/ofxOMXPlayer/archive/refs/tags/0.10.0.zip
unzip 0.10.0.zip 
cd ofxOMXPlayer-0.10.0/
sudo ./install_depends.sh
sudo apt-get install libavfilter-dev # no need because it already installed by install_depends.sh
./makeAllExamples.sh 
```

And get following output:

```
$ ./makeAllExamples.sh 
COMPILING PROJECT:  example-basic
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0 
with PKG_CONFIG_LIBDIR=
Compiling OF library for Release
make[1]: Entering directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0 
with PKG_CONFIG_LIBDIR=
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0 
with PKG_CONFIG_LIBDIR=
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0 
with PKG_CONFIG_LIBDIR=
Done!
make[1]: Leaving directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'

Compiling example-basic for Release
make[1]: Entering directory '/home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0 
with PKG_CONFIG_LIBDIR=
Compiling /home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic/src/ofApp.cpp
g++ -c -O3 -DNDEBUG -Wall -Werror=return-type -std=c++14 -DGCC_HAS_REGEX -march=armv6 -mfpu=vfp -mfloat-abi=hard -fPIC -ftree-vectorize -Wno-psabi -pipe -DOF_USING_GTK -DTARGET_RASPBERRY_PI -DSTANDALONE -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM  -I/opt/vc/include -I/opt/vc/include/IL -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -pthread -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/orc-0.4 -I/usr/include/gstreamer-1.0 -I/usr/include/AL -I/usr/include/arm-linux-gnueabihf -I/usr/include/rtaudio -I/usr/include/alsa -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/arm-linux-gnueabihf/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0 -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/uuid -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I../../../libs/glm/include -I../../../libs/glm/include/glm -I../../../libs/glm/include/glm/detail -I../../../libs/glm/include/glm/gtc -I../../../libs/glm/include/glm/gtx -I../../../libs/glm/include/glm/simd -I../../../libs/json/include -I../../../libs/kiss/include -I../../../libs/tess2/include -I../../../libs/utf8/include -I../../../libs/utf8/include/utf8 -I../../../libs/openFrameworks -I../../../libs/openFrameworks/types -I../../../libs/openFrameworks/sound -I../../../libs/openFrameworks/math -I../../../libs/openFrameworks/events -I../../../libs/openFrameworks/graphics -I../../../libs/openFrameworks/app -I../../../libs/openFrameworks/communication -I../../../libs/openFrameworks/utils -I../../../libs/openFrameworks/video -I../../../libs/openFrameworks/3d -I../../../libs/openFrameworks/gl -I/home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic/src -I/. -I. -I./src -I./obj -I./obj/linuxarmv6l -I./obj/linuxarmv6l/Release -I./obj/linuxarmv6l/Release/src -I/./bin -I./bin -I./bin/data -MMD -MP -MF obj/linuxarmv6l/Release/src/ofApp.d -MT obj/linuxarmv6l/Release/src/ofApp.o -o obj/linuxarmv6l/Release/src/ofApp.o -c /home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic/src/ofApp.cpp
In file included from /usr/include/boost/filesystem/path_traits.hpp:29,
                 from /usr/include/boost/filesystem/path.hpp:25,
                 from /usr/include/boost/filesystem.hpp:16,
                 from ../../../libs/openFrameworks/utils/ofFileUtils.h:20,
                 from ../../../libs/openFrameworks/ofMain.h:7,
                 from /home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic/src/ofApp.h:3,
                 from /home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-basic/src/ofApp.cpp:1:
/./bin/locale:1:1: error: stray ‘\177’ in program
```

And many errors below. It happens because there is `-I/./bin` in CFLAGS and compiler try to include /bin/locale instead of right locale header.

glitchcore, 2021-03-25:
Also I temporary rename /bin/locale to /bin/locale.1  and get next error:

```
In file included from /home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-direct-mode/src/ofApp.cpp:1:
/home/pi/openFrameworks/addons/ofxOMXPlayer-0.10.0/example-direct-mode/src/ofApp.h:4:10: fatal error: ofxOMXPlayer.h: No such file or directory
 #include "ofxOMXPlayer.h"
```

mmunderlinegithub, 2021-05-03:
Hey glitch core, which PI version are you using and which OS is installed?