jvcleave/ofxOMXPlayer #166: example compiling fatal error

mmunderlinegithub, 2021-04-11:
Hi everyone

I'm a complete noob but I managed to run the openFramworks examples on my pi 4. Quite proud of that already. 
But now I'd like to install the addon ofxOMXPlayer. The installer runs perfectly through but when I run "sudo ./makeAllExamples.sh". I get a lot of these error stray msgs. In the end the compilation fails with this message:

```
`In file included from /home/pi/openFrameworks/addons/ofxOMXPlayer-master/example-wrapper/src/ofApp.cpp:1:
/home/pi/openFrameworks/addons/ofxOMXPlayer-master/example-wrapper/src/ofApp.h:4:10: fatal error: ofRPIVideoPlayer.h: No such file or directory
 #include "ofRPIVideoPlayer.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:217: obj/linuxarmv6l/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxOMXPlayer-master/example-wrapper'
make: *** [/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:129: Release] Error 2
example-wrapper COMPILE FAIL
```

I set the GPU to 128 of memory to compile the examples... Any tips? I also ready the other issue regarding "ofRPIVideoPlayer.h" but I think it's a different problem.

Thanks for the help!
M

jvcleave, 2021-04-11:
Not sure about the ofRPIVideoPlayer.h error but be aware that this won't work on the RPI4

https://github.com/jvcleave/ofxOMXPlayer/issues/157

mmunderlinegithub, 2021-04-11:
Thanks for the fast answer! but only the texture mode as far I can understand... I only need ofxPiMapper running... Do you think this will run?

jvcleave, 2021-04-11:
unfortunately not :(

mmunderlinegithub, 2021-04-12:
ohhh so sad. but thanks for the kind help. I ordered already three rasberry 3s. Not gonna give up just because of this.

mmunderlinegithub, 2021-04-19:
Hey jvcleave.... sorry to interrupt you again. I just received my Raspberry Pi 3 (Model B + (ARMv8)). Which version of Rasbian would you recommend to install? The latest? Sorry for this noob questions....